rust-embedded/svdtools #9: RIIR

nickray, 2020-01-26:
There have been votes to rewrite this in Rust, I think most everyone would be open to this, but it would need an external initiative.

MarcoIeni, 2020-03-26:
Hi, I would like to try to work on this.
Should I fork the project into my github account and edit it or should I start a new project from scratch?
Of course any advice is appreciated :)

nickray, 2020-03-26:
Hey, cool, good luck! No preference from my side. If you do it in a fork of this repository (add Cargo.toml and src/ directory), might be easier to run comparison tests. I guess the main challenge will be to extract the implicit "API" or operations that `svd patch` implements, and model them explicitly.

MarcoIeni, 2020-03-28:
I forked this repo and I created [this](https://github.com/MarcoIeni/svdtools/tree/riir) branch.

Currently implemented tools:
- [x] interrupts (test missing)
- [ ] makedeps (work in progress)
- [ ] mmap
- [ ] patch

If you want to help write here or open an issue in my fork.
I think it is better to create a pull request only after everything is finished and refactored.

MarcoIeni, 2020-03-28:
The `interrupts` command was implemented by me, [crash-g](https://github.com/crash-g) and other people who I don't know the github handle during the [Open Source Saturday Italy](https://oss-italy.github.io/) :smile: 

[image]

jessebraham, 2020-03-28:
One problem I've encountered with the current implementation is that patching operations are order-dependent.

For example, you cannot `derive`/`copy` from a peripheral and then `delete` the base peripheral, as deletions are processed first. Changing the order of the operations in [`process_device`](https://github.com/stm32-rs/svdtools/blob/master/svdtools/patch.py#L959-L1005) can enable this behaviour, but in turn can cause patching to fail on other projects.

No obligation to address this now, just figured I'd bring it up because IMO it would help make the tool a bit more flexible. Thanks for taking a swing at this issue!

MarcoIeni, 2020-04-02:
ok, @jessebraham, I will keep that in mind :)

Anyway, I just found out that [svd-parser](https://crates.io/crates/svd-parser) exists. ^^'
I think I will use it for the `mmap` command for sure, but I am not sure if I should use for the `interrupts` one.
In fact, the `mmap` command parses all the svd, while the `interrupts` command parses only interrupts and peripheral's names.

For the `interrupts` command it is a waste of resources to read all the svd, therefore my custom [solution](https://github.com/MarcoIeni/svdtools/blob/riir/src/interrupts/svd_reader.rs) that only reads the needed fields is better in terms of performances but worse in terms of duplication of code.

I did a quick comparison with the `time` linux command and my solution is almost 15 times faster.
Even the current python solution is faster.

Should I keep my custom solution or use the `svd-parser` for the `interrupts` command?

MarcoIeni, 2020-04-15:
Anyone successfully wrote the [Device](https://github.com/rust-embedded/svd/blob/master/src/svd/device.rs) struct of [svd-parser](https://crates.io/crates/svd-parser) back to xml?
I tried with [quick-xml](https://docs.rs/quick-xml/0.18.1/quick_xml/) but serde unit variants is not yet supported in master. I tried with [this](https://github.com/Ten0/quick-xml) fork, but it doesn't seem to work as expected.
For example, instead of `<endian>little</endian>` I get `<Little />`

MarcoIeni, 2020-04-16:
I had a revelation ✨

Maybe we don't need to write the xml back.

At the moment with python we:
1. read the svd 
2. patch the svd with yaml content
3. write patched svd to file system
4. parse patched svd with [svd-parser](https://crates.io/crates/svd-parser) to produce the [Device](https://github.com/rust-embedded/svd/blob/master/src/svd/device.rs) struct

But with rust we could:
1. read the svd with `svd-parser` to produce the `Device` struct
2. edit the `Device` struct with yaml content

Am I right or the patched svd on the file system is used in some way?
My point is that since I think that the `Device` struct is what is used by svd2rust, we could provide this struct directly to svd2rust

adamgreig, 2020-04-16:
A lot of our other tooling uses the patched SVDs, for example generating the HTML pages and memory maps. It's also really useful to be able to read the patched SVD to see what you're getting.

MarcoIeni, 2020-04-16:
Ok :(
So my [previous question](https://github.com/stm32-rs/svdtools/issues/9#issuecomment-614306467) is still valid.

adamgreig, 2020-04-16:
I'm afraid I don't know much about svd-parser but it's worth opening an issue to see if there's interest in being able to write SVD files back out.

One worry I'd have about going through a `Device` struct is it might make round-tripping very hard, i.e. the output format could be totally different to the input, instead of only being different where there were changes, which makes it very hard to diff.

MarcoIeni, 2020-04-16:
Ok, I think I understood how to do it, they provide an [encode](https://github.com/rust-embedded/svd/blob/608088cd945f1c700176ff6b50b7a501fd5c7125/src/lib.rs#L64-L65) method which is not listed in their docs because you have to enable with the "unproven" feature, which I suppose it is a way to tell that this feature is still experimental.

> One worry I'd have about going through a Device struct is it might make round-tripping very hard, i.e. the output format could be totally different to the input, instead of only being different where there were changes, which makes it very hard to diff.

The output they provide is not formatted, so I will search for a way to format it in the way the initial svd is formatted in order to be able to diff the two.
I need something like [this](https://www.freeformatter.com/xml-formatter.html) but that runs locally.

MarcoIeni, 2020-04-16:
# simple format
I used [xmllint](http://xmlsoft.org/xmllint.html) as an xml formatter and I get nice indendentation.

The problem is that, as Adam said, I get small differences like this:
[image]

To the left there is the python output, to the right the output of my rust program (that uses svd_parser) after being formatted.
- displayName is not contained in the [RegisterInfo](https://github.com/rust-embedded/svd/blob/master/src/svd/registerinfo.rs).
- size is in decimal
- the items are in different order

I think that first two problems could be fixed easily in the svd_parser repo.
For the last one, we could first sort the xml nodes and then do the diff.

# format and sort
I sorted (and formatted) the two svd using a really simple python script.

This is how diffs look like now:
[image]

or:
[image]

As you can see now the differences are only different conventions (exadecimal vs decimal for example) or the missing displayName field.

# svd_parser vs dictionary
I took the direction of using the svd_parser crate, in order to avoid having duplicated code around, have better error handling and keep the code cleaner.

As I said, to have nicer diff with previous svd we could fix different conventions and add the missing fields.
Alternatively, instead of using the original svd for the diff, we could use the original svd togenerate an svd with svd_parser and use this generated file instead. In this way we diff two svd files generated by svd_parser, before and after the patch, therefore conventions are the same, so the diff produced by the patch is clear. 

However, I am still in time to change direction, I only implemented: include, modify device, copy peripheral and delete peripheral.
By using the python approach we could consider everything as a dictionary, therefore the diff between the two svd is simpler to see.

Which one of the two approaches do you prefer?

MarcoIeni, 2020-04-24:
# See the difference on your pc
In the following I show you how to evaluate on your pc the svd_parser approach.

This is the python script I used to format and sort an xml, I called it `xml_sort.py`:
```
from lxml import etree
import sys
import io

xml_path = sys.argv[1]
xml_out_path = sys.argv[2]

with io.open(xml_path, 'r', encoding='utf8') as f:
    data = f.read()

doc = etree.XML(data,etree.XMLParser(remove_blank_text=True, encoding='utf-8'))

for parent in doc.xpath('//*[./*]'): # Search for parent elements
    parent[:] = sorted(parent,key=lambda x: x.tag)

xml_object = etree.tostring(doc, pretty_print=True, xml_declaration=True, encoding='UTF-8')

with open(xml_out_path, "wb") as writter:
    writter.write(xml_object)
```

1. See test inputs in directory `res/example1`
2. `cargo test` and see test outputs in directory `res/example1`
3. delete `<?xml version="1.0" encoding="UTF-8"?>` from the first line of stm32l4x2.svd.patched otherwise my xml_sort.py complains
4. format and sort patched svd:
    `python3 ~/path/to/xml_sort.py stm32l4x2.svd.patched stm32l4x2_ord.svd.patched`
5. diff against python output.
    `diff expected.svd stm32l4x2_ord.svd.patched`
6.  The result of the previous command is not good enough, so I sort the python output
     `python3 ~/path/to/xml_sort.py expected.svd expected_ord.svd`
7. diff against sorted python output.
    `diff expected_ord.svd stm32l4x2_ord.svd.patched`

I used [meld](https://meldmerge.org/) instead of diff.

You can consider this as a first demo :)

MarcoIeni, 2020-12-27:
Hi everyone!
This issue was abandoned, mainly because `svd patch` is pretty complex.
Anyway, the other subcommands were translated correctly to rust, so if you want we can substitute them with the python version.
If you have time to follow me in the process tell me in which way should I proceed (create a new repo?) and I will review the old code I've written and submit it.

burrbull, 2021-04-28:
@MarcoIeni if you are interested yet, see this `svd-parser` branch:
https://github.com/rust-embedded/svd/tree/resort

closed 2023-03-28 by burrbull