rust-embedded/svdtools #14: Patching "DEVù

aurelj, 2020-02-21:


aurelj, 2020-02-21:
Commit 3052b8b broke patching of generic device like "DEV*" when one of the device is derivedFrom another one. This is very common in stm32-rs.

Here is an example of traceback (SPI2 is derived from SPI1 and CR1 is patched on SPI*):
```
Traceback (most recent call last):
  File "/home/aurel/.local/bin/svd", line 10, in <module>
    sys.exit(svdtools_cli())
  File "/home/aurel/.local/lib/python3.7/site-packages/click/core.py", line 764,
 in __call__
    return self.main(*args, **kwargs)
  File "/home/aurel/.local/lib/python3.7/site-packages/click/core.py", line 717,
 in main
    rv = self.invoke(ctx)
  File "/home/aurel/.local/lib/python3.7/site-packages/click/core.py", line 1137
, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/aurel/.local/lib/python3.7/site-packages/click/core.py", line 956,
 in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/aurel/.local/lib/python3.7/site-packages/click/core.py", line 555,
 in invoke
    return callback(*args, **kwargs)
  File "/home/aurel/.local/lib/python3.7/site-packages/svdtools/cli.py", line 16
, in patch
    svdtools.patch.main(yaml_file)
  File "/home/aurel/.local/lib/python3.7/site-packages/svdtools/patch.py", line
1010, in main
    process_device(svd, root)
  File "/home/aurel/.local/lib/python3.7/site-packages/svdtools/patch.py", line
990, in process_device
    d.process_peripheral(periphspec, device[periphspec], update_fields)
  File "/home/aurel/.local/lib/python3.7/site-packages/svdtools/patch.py", line
419, in process_peripheral
    p.process_register(rspec, register, update_fields)
  File "/home/aurel/.local/lib/python3.7/site-packages/svdtools/patch.py", line
760, in process_register
    raise MissingRegisterError("Could not find {}:{}".format(pname, rspec))
svdtools.patch.MissingRegisterError: Could not find SPI2:CR1
make: *** [Makefile:39: svd/stm32f0x0.svd.patched] Error 1
```

See also my comment on #13

nickray, 2020-02-23:
I reverted said change and released v0.1.4 without it.
PR https://github.com/stm32-rs/svdtools/pull/15 should prevent this in the future.

closed 2020-02-23 by nickray