rust-embedded/svdtools #23: patch testing

MarcoIeni, 2020-04-07:
I am trying to solve #9 and I noticed that the `patch` command has no tests.
If you want to help me you can provide here some test cases 😀

You should provide: test description, input svd and yaml.

You don't have to provide also the patched svd (the output), since I can obtain it by launching the python script.

## Example
### Test description
Modify register field.

### svd
```xml
<?xml version="1.0" encoding="utf-8" standalone="no"?>
<device schemaVersion="1.1"
xmlns:xs="http://www.w3.org/2001/XMLSchema-instance"
xs:noNamespaceSchemaLocation="CMSIS-SVD_Schema_1_1.xsd">
  <name>STM32L4x2</name>
  <version>1.6</version>
  <description>STM32L4x2</description>
  <!-- details about the cpu embedded in the device -->
  <cpu>
    <name>CM4</name>
    <revision>r1p0</revision>
    <endian>little</endian>
    <mpuPresent>false</mpuPresent>
    <fpuPresent>false</fpuPresent>
    <nvicPrioBits>3</nvicPrioBits>
    <vendorSystickConfig>false</vendorSystickConfig>
  </cpu>
  <!--Bus Interface Properties-->
  <!--Cortex-M3 is byte addressable-->
  <addressUnitBits>8</addressUnitBits>
  <!--the maximum data bit width accessible within a single transfer-->
  <width>32</width>
  <!--Register Default Properties-->
  <size>0x20</size>
  <resetValue>0x0</resetValue>
  <resetMask>0xFFFFFFFF</resetMask>
  <peripherals>
    <peripheral>
      <name>DAC1</name>
      <description>Digital-to-analog converter</description>
      <groupName>DAC</groupName>
      <baseAddress>0x40007400</baseAddress>
      <addressBlock>
        <offset>0x0</offset>
        <size>0x400</size>
        <usage>registers</usage>
      </addressBlock>
      <registers>
        <register>
          <name>CR</name>
          <displayName>CR</displayName>
          <description>control register</description>
          <addressOffset>0x0</addressOffset>
          <size>0x20</size>
          <access>read-write</access>
          <resetValue>0x00000000</resetValue>
          <fields>
            <field>
              <name>EN1</name>
              <description>DAC channel1 enable</description>
              <bitOffset>0</bitOffset>
              <bitWidth>1</bitWidth>
            </field>
          </fields>
        </register>
      </registers>
    </peripheral>
  </peripherals>
</device>
```

### yaml
```yaml
_svd: stm32l4x2.svd

# SVD incorrectly labels APB1ENR1 bit 18 as USART1EN instead of USART3EN.
# SVD incorrectly labels APB1ENR1 bit 26 as USBF instead of USBFSEN.
DAC1:
  CR:
    _modify:
      EN1:
        name: EN2
        description: EN2 description
```

nickray, 2020-04-09:
This is probably also a good way to document the more complex operations 😅

MarcoIeni, 2020-04-09:
I am specifying the structure of the yaml files [here](https://github.com/MarcoIeni/svdtools/blob/riir/src/patch/yaml_parser.rs). Of course is a work in progress.

> This is probably also a good way to document the more complex operations.

Yes, this could solve also #5

> I personally have difficulty using the "API" as it's mostly implicitly implementation-defined. Would be awesome to specify it in a structured way, and pick tests according to that.

Yes, if somebody who has experience with this tool does it it would be awesome.
Maybe here we can define a general structure for each of the patch commands.

For example something like this could be created for each patch command:

# include
`_include` let you specify other yaml file that will be merged with this one.
`_include` can be only specified at the root of the yaml file or at the peripheral level.

```
PERIPH:
  _include:
    - path/to/peripheral_patch.yaml

_include:
  - path/to/other_patch.yaml
```

## Example
See [first](https://github.com/stm32-rs/svdtools/issues/23#issue-596178258) post for an example.