rust-embedded/svdtools #30: Cannot copy from a peripheral that is also deleted in the same patch

ijager, 2020-07-22:
If I try to first `_copy` (or `_rebase`) a peripheral and then delete the original it does not work. This is the most simplest reproduction of this issue:

```xml
_svd: svd/stm32g070.svd

_copy:
  TIM3:
    from: TIM2

_delete:
  - TIM2
```

It crashes with the error 
```
svdtools.patch.SvdPatchError: peripheral None not found
```

How is the order of operations defined? It seems like `_delete` is done before `_copy`. If I remove the `_delete`,  the `_copy` works fine. Is there a workaround?

duskmoon314, 2022-03-19:
I'm facing the same issue.

It seems that the order is hardcoded in `src/patch/peripheral.rs` of the rust version.

Have you figured out how to workaround?

ijager, 2022-03-19:
Looks like I renamed TIM3 to TIM2:

```yaml
_modify:
  name: STM32G070

  # Remove TIM3 and rename TIM2 to TIM3. Because TIM3 was derive from TIM2
  TIM2:
    name: TIM3
    baseAddress: "0x40000400"
```