rust-embedded/svdtools #41: Conflict when adding enumeratedValues to fields in registers in a cluster

BryanKadzban, 2020-10-04:
The Atmel SAML10 chips have a somewhat odd structure for their UART/SPI/I2C peripherals.  There are a couple of single SERCOM peripherals, which can be configured into one of those modes based on a MODE register, whose offset is the same for all modes.

The Atmel-provided SVD files model this like so:

<peripheral>
  <name>SERCOM0</name>
  <registers>
    <cluster><name>I2CM</name>
      <register>
        <name>CTRLA</name>
        <fields>
          <field><name>SDAHOLD</name></field>
          ....
        </fields>
      </register>
      ...
    </cluster>
    <cluster><name>SPI</name>
      <register>
        <name>CTRLA</name>
        <fields>
          <field><name>CPHA</name></field>
          ....
        </fields>
      </register>
    </cluster>
  </registers>
</peripheral>

That is, there's a cluster for each independent configuration of the peripheral, with the same register name defined, and different fields defined.  (SDAHOLD does not exist in SPI mode, and CPHA does not exist in I2C mode.)  Other fields, such as MODE, are in both.

I don't see a way to select just one of these clusters to modify.  If I match on CTRLA then I can't match on the right fields, because the selector code in patch.py returns all four of the CTRLA register tags for all four clusters.  If I try to introduce a _cluster: entry in the YAML then the code insists on grouping all four of the CTRLA registers together into a block, and failing to validate the offsets or the bitmasks.

Is there a way to use the cluster name purely for selection, instead of for cluster creation?

BryanKadzban, 2020-10-05:
I've changed a local copy of patch.py to allow this (by adding a _clustermatch: tag in the YAML and copying a lot of code).  I can try to get a PR together if people would find it useful in this repo.  Unless I'm just doing something very wrong in the YAML I've tried with the stock python code.

burrbull, 2023-03-28:
Should be possible now by _clusters tag

closed 2023-03-28 by burrbull