rust-embedded/svdtools #60: Having an enumerated value `Off` throws an error

jorgeig-space, 2021-05-29:
I am writing a peripheral for STM32WLE. One of the bits is best described as "Off", so I wrote:

```
    RRS:
      Off: [0, "SRAM2 powered off in Standby mode (SRAM2 content lost)"]
      OnLPR: [1, "SRAM2 powered by the low-power regulator in Standby mode (SRAM2 content kept)"]
```

When running make, I get the following output:
```
❯ make
svd makedeps devices/stm32wle5.yaml .deps/stm32wle5.d
svd patch devices/stm32wle5.yaml
Traceback (most recent call last):
  File "/Users/xxxx/.pyenv/versions/stm32-rs/bin/svd", line 8, in <module>
    sys.exit(svdtools_cli())
  File "/Users/xxxx/.pyenv/versions/3.8.9/envs/stm32-rs/lib/python3.8/site-packages/click/core.py", line 1137, in __call__
    return self.main(*args, **kwargs)
  File "/Users/xxxx/.pyenv/versions/3.8.9/envs/stm32-rs/lib/python3.8/site-packages/click/core.py", line 1062, in main
    rv = self.invoke(ctx)
  File "/Users/xxxx/.pyenv/versions/3.8.9/envs/stm32-rs/lib/python3.8/site-packages/click/core.py", line 1668, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/xxxx/.pyenv/versions/3.8.9/envs/stm32-rs/lib/python3.8/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/xxxx/.pyenv/versions/3.8.9/envs/stm32-rs/lib/python3.8/site-packages/click/core.py", line 763, in invoke
    return __callback(*args, **kwargs)
  File "/Users/xxxx/.pyenv/versions/3.8.9/envs/stm32-rs/lib/python3.8/site-packages/svdtools/cli.py", line 16, in patch
    svdtools.patch.main(yaml_file)
  File "/Users/xxxx/.pyenv/versions/3.8.9/envs/stm32-rs/lib/python3.8/site-packages/svdtools/patch.py", line 1407, in main
    process_device(svd, root)
  File "/Users/xxxx/.pyenv/versions/3.8.9/envs/stm32-rs/lib/python3.8/site-packages/svdtools/patch.py", line 1384, in process_device
    d.process_peripheral(periphspec, device[periphspec], update_fields)
  File "/Users/xxxx/.pyenv/versions/3.8.9/envs/stm32-rs/lib/python3.8/site-packages/svdtools/patch.py", line 661, in process_peripheral
    p.process_register(rspec, register, update_fields)
  File "/Users/xxxx/.pyenv/versions/3.8.9/envs/stm32-rs/lib/python3.8/site-packages/svdtools/patch.py", line 1032, in process_register
    r.process_field(pname, fspec, field)
  File "/Users/xxxx/.pyenv/versions/3.8.9/envs/stm32-rs/lib/python3.8/site-packages/svdtools/patch.py", line 1234, in process_field
    self.process_field_enum(pname, fspec, field)
  File "/Users/xxxx/.pyenv/versions/3.8.9/envs/stm32-rs/lib/python3.8/site-packages/svdtools/patch.py", line 1260, in process_field_enum
    enum = make_enumerated_values(name, field, usage=usage)
  File "/Users/xxxx/.pyenv/versions/3.8.9/envs/stm32-rs/lib/python3.8/site-packages/svdtools/patch.py", line 148, in make_enumerated_values
    if vname.startswith("_"):
AttributeError: 'bool' object has no attribute 'startswith'
make: *** [svd/stm32wle5.svd.patched] Error 1
```

I am using Python 3.8.9, compiled on x86_64 (I have an M1 so I tried both x86_64 and aarch64).

If instead, I change the bit to:
```
    RRS:
      PowerOff: [0, "SRAM2 powered off in Standby mode (SRAM2 content lost)"]
      OnLPR: [1, "SRAM2 powered by the low-power regulator in Standby mode (SRAM2 content kept)"]
```

Everything works normally.

I have tried in other peripherals and bits and the same issue occurs. The same is true if the value is `On`, e.g.:
```
    RRS:
      PowerOff: [0, "SRAM2 powered off in Standby mode (SRAM2 content lost)"]
      On: [1, "SRAM2 powered by the low-power regulator in Standby mode (SRAM2 content kept)"]
```

adamgreig, 2021-05-29:
This is an infuriating YAML issue where On and Off are treated as special Boolean values and turned into Python True and False instead of strings. It should work OK if you put "On" and "Off" (inside quotation marks) though.

jorgeig-space, 2021-05-29:
Oh wow, I did not know that about YAML. Thanks for clarifying.

In that case, I will close the issue. Do you mind if I write this somewhere in the README and do a PR?

adamgreig, 2021-05-29:
Sure, thank you!

closed 2021-05-29 by jorgeig-space