rust-embedded/svdtools #62: Allow adding derivedFrom attribute to enumeratedValues

qwandor, 2021-06-28:
As far as I can tell, there is no way to use the svd patch tool to add a `derivedFrom` attribute to an `enumeratedValues` tag. This should be possible. I would like to do this so that `svd2rust` uses the same enum type for the equivalent field on several different peripherals. I would expect to be able to do something like

```yaml
TIMER2:
  CHCTL2:
    _derive:
      CH0P: TIMER0.CHCTL2.CH0P
```

Or perhaps:

```yaml
TIMER2:
  CHCTL2:
    _modify:
      CH0P:
        derivedFrom: TIMER0.CHCTL2.CH0P
```

qwandor, 2021-06-29:
It turns out this is already possible, it's just not documented and I had to read the source to figure it out. I was able to do what I what I wanted like:

```yaml
TIMER2:
  CHCTL2:
    CH0P:
      _derivedFrom: "TIMER0.CHCTL2.CH0P.CH0P"
```

So it would be good to document this. It's also not clear why the field name has to be repeated twice.

burrbull, 2021-06-30:
> It's also not clear why the field name has to be repeated twice.

First is name of field. Second is name of enumeratedValues.

kossnikita, 2023-09-27:
Is it possible to use it for separate _read and _write?

closed 2023-12-18 by burrbull