rust-embedded/svdtools #100: conversion error: unsure if I'm just missing something

hanetzer, 2022-04-11:
Ello. I'm using the rust svdtools program to build an svd for a chip without one.
I started by converting the rp2040.svd to rp2040.yaml as a reference, and began
writing the yaml for my chip, to convert to svd later for use with gdb and such.
Some of the data is dummied out (for instance, its not a traditional mcu, but is
a cortex-a7 proc). Attempting to convert with what I currently have (below) results
in the following error:
`[2022-04-11T13:49:24Z ERROR svdtools::cli] peripherals[0]: invalid type: map, expected a sequence at line 20 column 9`
Line 20 column nine translates to the `:` before BOOT_ADDR; this seems correct,
as the rp2040.yaml file has the same kind of setup:

Am I missing something or is this a legit bug?
```yaml
---
vendor: Raspberry Pi
name: RP2040
version: "0.1"
description: ""
licenseText: "\n    Copyright (c) 2020 Raspberry Pi (Trading) Ltd. \\n\n    \\n\n    SPDX-License-Identifier: BSD-3-Clause\n  "
cpu:
  name: CM0PLUS
  revision: r0p1
  endian: little
  mpuPresent: true
  fpuPresent: false
  vtorPresent: true
  nvicPrioBits: 2
  vendorSystickConfig: false
  deviceNumInterrupts: 26
addressUnitBits: 8
width: 32
peripherals:
  - name: XIP_CTRL
```
```yaml
---
vendor: Hisilicon
name: Hi3521A
version:  "0.0"
description: ""
licenseText: ""
cpu:
  name: other
  revision: r0p1
  endian: little
  mpuPresent: true
  fpuPresent: true
  vtorPresent: true
  nvicPrioBits: 0
  vendorSystickConfig: false
  deviceNumInterrupts: 64
addressUnitBits: 8
width: 32
peripherals:
  - name: BOOT_ADDR
    version: "1"
    description: ""
    baseAddress: 0x0
    size: 32
    addressBlock:
      offset: 0
      size: 0x4000000
      usage: buffer
  - name: BOOT_ROM
    version: "1"
    description: ""
    baseAddress: 0x04000000
    size: 32
    addressBlock:
      offset: 0
      size: 0x1000
[... 493 lines cut ...]
    baseAddress: 0x13150000
    size: 32
    addressBlock:
      offset: 0
      size: 0x10000
      usage: registers
  - name: SFC_MEM
    version: "1"
    description: ""
    baseAddress: 0x14000000
    size: 32
    addressBlock:
      offset: 0
      size: 0x1000000
      usage: buffer
  - name: DRAM
    version: "1"
    description: ""
    baseAddress: 0x80000000
    size: 32
    addressBlock:
      offset: 0
      size: 0x80000000
      usage: buffer
```

hanetzer, 2022-04-11:
Actually, I found a conversion issue. When converting a yaml to svd, the resultant svd file does
not contain the schema information some svd tools (in my case, [SVD-Loader-Ghidra](https://github.com/leveldown-security/SVD-Loader-Ghidra)) require for
operation.
Converted SVD:
```xml
<device schemaVersion="" xmlns:xs="" xs:noNamespaceSchemaLocation="">
```

'real' SVD:
```xml
<device schemaVersion="1.1" xmlns:xs="http://www.w3.org/2001/XMLSchema-instance" xs:noNamespaceSchemaLocation="CMSIS-SVD.xsd">
```

burrbull, 2022-04-12:
#102

hanetzer, 2022-04-13:
Yep, that fixed that. Thanks for the info.

closed 2022-04-13 by hanetzer