rust-embedded/svdtools #126: svdtools: patching yaml files/multipart yaml conversion?

hanetzer, 2022-09-08:
So, I know this is *mostly* used to patch vendor-shipped svd files, but as prior issues have stated,
I've been using this to write svd's for cortex-a socs in yaml for use in other projects (ghidra, pycortexmdebug,
etc).

Is it possible, for example, do do the patching type stuff without a 'real' 'seed' svd file?
Like, I'd want to define all the svd blocks for stuff that is unique to this particular soc (clock controllers,
system controllers, various glue peripherals) in the primary yaml file, and draw from various 'library'
yaml files that define 'standard' peripherals (arm primecell stuff, designware stuff, etc).

I realize I can just concatenate main.yaml, gpioX.yaml, and so on, but that gets awkward as time goes on
and the file grows.

For example, one could have a file tree like this:
```
.
├── hi3516dv300.yaml
├── hi3521av100.yaml
└── peripherals
    ├── designware
    │   └── dw-mmc.yaml
    └── primecell
        ├── pl011.yaml
        ├── pl061.yaml
        └── sp804.yaml
```
Where hi3516dv300.yaml defines that there is a pl061 gpio at every `0x120d_{0..b}000` address block but
hi3521av100 has them at `0x12{15..22}_0000` and so on?

hanetzer, 2022-09-08:
hmm... maybe do that thing you guys did with "GPIO*" inclusion, and only include the baremost data
in the 'seed' svd to set base address and interrupt?

burrbull, 2022-09-09:
`svdtools patch` was created and optimized for patching existent files.
I don't think it is good idea to use same syntax for other tasks like creating new files.
There are alternatives you could use:
1. `svd-rs` supports `serde`. So you can create SVD-like files in yaml (json) syntax if you don't like XML and then convert them with `svdtools convert` if it is needed. So you can write wrapper which merge peripheral files in one SVD-like and then `serde` it.
2. Use non-SVD alternatives like `embassy` metapack.

hanetzer, 2022-09-09:
Another idea: have the same syntax for the primary patch yaml file (the one that includes the svd file), but allow
for using a yaml file as the 'seed' svd?

flaviojs, 2026-03-13:
In my case I'm generating a svd from yaml with the convert command according to a spec, but now I want to apply patches without needed an intermediate svd file.

I made a PR that adds support for yaml/json files in "_svd", which also fixes this.