rust-embedded/svdtools #139: makedeps fails to generate valid `.d` files on windows.

hydra, 2023-03-22:
There is a failing test:

[image]

```
Left:  test.d: \\?\D:\Users\Hydra\Documents\dev\projects\stm32-rs\svdtools\res\makedeps\sub-tests\inc1.yaml \\?\D:\Users\Hydra\Documents\dev\projects\stm32-rs\svdtools\res\makedeps\sub-tests\inc2.yaml
Right: test.d: D:\Users\Hydra\Documents\dev\projects\stm32-rs\svdtools\res\makedeps\sub-tests/inc1.yaml D:\Users\Hydra\Documents\dev\projects\stm32-rs\svdtools\res\makedeps\sub-tests/inc2.yaml
```

The use of `std::fs::canonicalize` is adding `\\?\` in addition to removing `\.\` path components.

These paths are not valid paths when `svdtools` is used as part of `stm32-rs` when running `make patch` in cygwin.
The resulting error from the command that uses the generated dependencies `make svd2rust` is as follows:

```
$ make svd2rust
.deps/stm32c011.d:1: *** target pattern contains no '%'.  Stop.
```

If you inspect the generated file you see this:
```
$ more .deps/stm32c011.d
stm32c011.d: \\?\D:\Users\Hydra\Documents\dev\projects\stm32-rs\stm32-rs\devices\common_patches\c0.yaml \\?\D:\Users\Hydra\Documents\dev\projects\stm32-rs\stm32-rs\devices\common_patches\nvic\2_prio_bits.yaml
```

As a workaround I hacked up `abspath` as follows:

```
/// Gets the absolute path of relpath from the point of view of frompath.
fn abspath(frompath: &Path, relpath: &Path) -> Result<PathBuf, std::io::Error> {
    //std::fs::canonicalize(frompath.parent().unwrap().join(relpath))
    let path = frompath.parent().unwrap().join(relpath);

    let path_str = format!("{}", path.display()).replace("\\", "/");

    Result::<_,std::io::Error>::Ok(PathBuf::from(path_str))
}
```

and now the test fails as follows:

```
Left:  test.d: D:/Users/Hydra/Documents/dev/projects/stm32-rs/svdtools/res/makedeps/sub-tests/inc1.yaml D:/Users/Hydra/Documents/dev/projects/stm32-rs/svdtools/res/makedeps/sub-tests/inc2.yaml
Right: test.d: D:\Users\Hydra\Documents\dev\projects\stm32-rs\svdtools\res\makedeps\sub-tests/inc1.yaml D:\Users\Hydra\Documents\dev\projects\stm32-rs\svdtools\res\makedeps\sub-tests/inc2.yaml
```

However, the resulting `.d` files are now valid, here's the same one from the stm32-rs project from before:

```
$ more .deps/stm32c011.d
stm32c011.d: devices/./common_patches/c0.yaml devices/./common_patches/nvic/2_prio_bits.yaml
```

I'm no expert on the String, PathBuf or FS APIs so not sure what a proper cross-platform fix is, but this unblocked me today.

burrbull, 2023-03-26:
@hydra Could you check #140 ?

hydra, 2023-03-27:
@burrbull yes, I will give that a whirl ASAP and report back.

burrbull, 2023-03-28:
Reopen if still not solved

hydra, 2023-03-28:
```
$ RUST_BACKTRACE=1 cargo test
    Finished test [unoptimized + debuginfo] target(s) in 0.08s
     Running unittests src\lib.rs (target\debug\deps\svdtools-79fa8ce3a1e1d116.exe)

running 9 tests
test common::str_utils::tests::correctly_format_address ... ok
test interrupts::interrupt_list::tests::correct_gaps_if_zero_and_two_are_missing ... ok
test interrupts::svd_reader::tests::peripherals_interrupts_are_parsed_correctly ... ok
test mmap::mmap_cli::tests::mmap ... ok
test patch::device::tests::add_peripherals ... ok
test patch::device::tests::copy_peripherals ... ok
test patch::device::tests::modify_device ... ok
test patch::device::tests::delete_peripherals ... ok
test makedeps::makedeps_cli::tests::test_makedeps ... FAILED

failures:

---- makedeps::makedeps_cli::tests::test_makedeps stdout ----
thread 'makedeps::makedeps_cli::tests::test_makedeps' panicked at 'assertion failed: `(left == right)`
  left: `"C:\\cygwin64\\tmp\\.tmpbghyq2\\test.d: \\\\?\\D:\\Users\\Hydra\\Documents\\dev\\projects\\stm32-rs\\svdtools\\res\\makedeps\\sub-tests\\inc1.yaml \\\\?\\D:\\Users\\Hydra\\Documents\\dev\\projects\\stm32-rs\\svdtools\\res\\makedeps\\sub-tests\\inc2.yaml\n"`,
 right: `"C:\\cygwin64\\tmp\\.tmpbghyq2/test.d: D:\\Users\\Hydra\\Documents\\dev\\projects\\stm32-rs\\svdtools\\res\\makedeps\\sub-tests/inc1.yaml D:\\Users\\Hydra\\Documents\\dev\\projects\\stm32-rs\\svdtools\\res\\makedeps\\sub-tests/inc2.yaml\n"`', src\makedeps\makedeps_cli.rs:80:9
stack backtrace:
   0: rust_begin_unwind
             at /rustc/065852def0903296da33a9eaf557f230bcf3a61a/library\std\src/panicking.rs:575:5
   1: core::panicking::panic_fmt
             at /rustc/065852def0903296da33a9eaf557f230bcf3a61a/library\core\src/panicking.rs:64:14
   2: core::panicking::assert_failed_inner
   3: core::panicking::assert_failed
             at /rustc/065852def0903296da33a9eaf557f230bcf3a61a\library\core\src/panicking.rs:211:5
   4: svdtools::makedeps::makedeps_cli::tests::test_makedeps
             at .\src\makedeps\makedeps_cli.rs:80:9
   5: svdtools::makedeps::makedeps_cli::tests::test_makedeps::{{closure}}
             at .\src\makedeps\makedeps_cli.rs:62:27
   6: core::ops::function::FnOnce::call_once
             at /rustc/065852def0903296da33a9eaf557f230bcf3a61a\library\core\src\ops/function.rs:250:5
   7: core::ops::function::FnOnce::call_once
             at /rustc/065852def0903296da33a9eaf557f230bcf3a61a/library\core\src\ops/function.rs:250:5
note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
```

@burrbull test still fails, as above

burrbull, 2023-03-28:
test fails, you are right.
But have you tried to run `svdtools patch`?
Also have you tried precompilled release version: https://github.com/stm32-rs/svdtools/releases/download/v0.3.0/svdtools-x86_64-pc-windows-msvc.zip ?

burrbull, 2023-03-28:
> ```
> $ more .deps/stm32c011.d
> stm32c011.d: \\?\D:\Users\Hydra\Documents\dev\projects\stm32-rs\stm32-rs\devices\common_patches\c0.yaml \\?\D:\Users\Hydra\Documents\dev\projects\stm32-rs\stm32-rs\devices\common_patches\nvic\2_prio_bits.yaml
> ```
Those are correct windows pathes.

hydra, 2023-03-29:
> Those are correct windows pathes

yes, they are _windows_ paths, but I'm running under cygwin, and cygwin `make` wants cygwin paths.

> But have you tried to run `svdtools patch`?

no, since I know it will fail as the same production code used by the tests will cause the paths to be generated in the same way.

> Also have you tried precompilled release version: https://github.com/stm32-rs/svdtools/releases/download/v0.3.0/svdtools-x86_64-pc-windows-msvc.zip ?

No, because past experience has taught me not to mix tools from different ecosystems.   WSL use WSL binaries, cygwin - use cygwin binaries, windows/vscode/powershell/cmdline - use windows binaries.

hydra, 2023-03-29:
IMHO the code used to generate paths needs to detect cygwin, and generate cygwin paths, not windows paths.

Cygwin paths are the same as linux paths.

Cygwin detection is usually done by running `uname` and looking for the `CYGWIN` string.

```
$ uname
CYGWIN_NT-10.0-19045
```

It might also be possible to supply a `--cygwin` option to the tool, to inform it to generate cygwin/linux style paths instead of windows paths if cygwin detection is done in the `Makefile` via `uname`.

A different approach could be to run the generated .d files through a processor which applies `cygpath` to each path.

```
$ cygpath --help
Usage: cygpath (-d|-m|-u|-w|-t TYPE) [-f FILE] [OPTION]... NAME...
       cygpath [-c HANDLE]
       cygpath [-ADHOPSW]
       cygpath [-F ID]

Convert Unix and Windows format paths, or output system path information

Output type options:

  -d, --dos             print DOS (short) form of NAMEs (C:\PROGRA~1\)
  -m, --mixed           like --windows, but with regular slashes (C:/WINNT)
  -M, --mode            report on mode of file (binmode or textmode)
  -u, --unix            (default) print Unix form of NAMEs (/cygdrive/c/winnt)
  -w, --windows         print Windows form of NAMEs (C:\WINNT)
  -t, --type TYPE       print TYPE form: 'dos', 'mixed', 'unix', or 'windows'

Path conversion options:

  -a, --absolute        output absolute path
  -l, --long-name       print Windows long form of NAMEs (with -w, -m only)
  -p, --path            NAME is a PATH list (i.e., '/bin:/usr/bin')
  -U, --proc-cygdrive   Emit /proc/cygdrive path instead of cygdrive prefix
                        when converting Windows path to UNIX path.
  -s, --short-name      print DOS (short) form of NAMEs (with -w, -m only)
  -C, --codepage CP     print DOS, Windows, or mixed pathname in Windows
                        codepage CP.  CP can be a numeric codepage identifier,
                        or one of the reserved words ANSI, OEM, or UTF8.
                        If this option is missing, cygpath defaults to the
                        character set defined by the current locale.

System information:

  -A, --allusers        use `All Users' instead of current user for -D, -O, -P
  -D, --desktop         output `Desktop' directory and exit
  -H, --homeroot        output `Profiles' directory (home root) and exit
  -O, --mydocs          output `My Documents' directory and exit
  -P, --smprograms      output Start Menu `Programs' directory and exit
  -S, --sysdir          output system directory and exit
  -W, --windir          output `Windows' directory and exit
  -F, --folder ID       output special folder with numeric ID and exit

Other options:

  -f, --file FILE       read FILE for input; use - to read from STDIN
  -o, --option          read options from FILE as well (for use with --file)
  -c, --close HANDLE    close HANDLE (for use in captured process)
  -i, --ignore          ignore missing argument
  -h, --help            output usage information and exit
  -V, --version         output version information and exit
```

My preference would be for the former, not the latter.

burrbull, 2023-12-18:
`cargo make` works fine on native windows. I don't plan to support `cygwin`.

PR #140 (2023-03-27): normpath
potentially fixes #139
closed 2023-12-18 by burrbull