rust-embedded/svdtools #146: svdtools does not handle derivin registers from another peripheral

ithinuel, 2023-04-01:
Example:

```yaml
PADS_QSPI:
  _delete:
    "VOLTAGE_SELECT":
  _derive:
    "VOLTAGE_SELECT":
      _from: "PADS_BANK0.VOLTAGE_SELECT"
```

fails with `svdtools.patch.SvdPatchError: register PADS_BANK0.VOLTAGE_SELECT not found`

Could be related to [`def derive_from`](https://github.com/stm32-rs/svdtools/blob/master/svdtools/patch.py#L888-L902) searching for the target register in the parent only rather than decomposing the target register's name.

burrbull, 2023-04-02:
All we can do here is disabling search for complicated paths.

Try https://crates.io/crates/svdtools instead

ithinuel, 2023-04-02:
Thank you for the feedback.
The rust implementation returns similar results:
```
[2023-04-02T09:27:51Z ERROR svdtools::cli] by svdtools (0.3.0)
    
    Caused by:
        0: Processing device `RP2040`
        1: According to `PADS_QSPI`
        2: Processing peripheral `PADS_QSPI`
        3: Deriving register `VOLTAGE_SELECT` from `Hash({String("_from"): String("PADS_BANK0.VOLTAGE_SELECT")})`
        4: register PADS_BANK0.VOLTAGE_SELECT not found
```

There cannot be "complicated path" as according to the SVD specs:

> Specify the register name from which to inherit data. Elements specified subsequently override inherited values. 
> **Usage:**
> - Always use the full qualifying path, which must start with the peripheral `<name>`, when deriving from another scope. (for example, in periperhal `B`, derive from `peripheralA.registerX`.
> - You can use the register `<name>` only when both registers are in the same scope.
> - No relative paths will work. 
>
> **Remarks:** When deriving, it is mandatory to specify at least the `<name>`, the `<description>`, and the `<addressOffset>`.

A register is either in the current scope or using an absolute path.

If I can find some time, I'll try to come we with a proposal to fix that. (I recon it may not be easy due to how iter muts are used).

ithinuel, 2023-04-02:
So, I spent the day trying to get something working.

I made some progress (a really tiny bit) on `svdtools` but, although `svd2rust` does handle the paths correctly, it does not handle `derivedFrom` very well.
So even if `svdtools` did process these, we couldn't do much beyond that.

I'm closing this, at least it'll be logged if someone tries to do something similar in the future.

closed 2023-04-02 by ithinuel