rust-embedded/svdtools #149: svd2rust is unable to generate pac for nxp svd file

felipebalbi, 2023-08-01:
Hi,

I'm attempting to generate a pac for an nxp rt633s device (svd: https://github.com/nxp-mcuxpresso/mcux-soc-svd/blob/main/MIMXRT633S/MIMXRT633S.xml) however, `svd2rust` ends with an error:

```
[INFO  svd2rust] Parsing device from SVD file
[INFO  svd2rust] Rendering device
[ERROR svd2rust] Error rendering device

    Caused by:
        0: Rendering error at peripheral
           Name: I2S0
           Description: LPC-Next0 I2S interface
           Group: I2S
        1: Could not expand register or cluster block
        2: Error expanding cluster
           Name: SECCHANNEL[%s]
           Description: no description available
        3: Cluster SECCHANNEL[%s] has size 24928 bits that is more then array increment 256 bits
```

Looking at that particular cluster, we have 96 bits (3x 32-bit registers):

```xml
        <cluster>
          <dim>3</dim>
          <dimIncrement>0x20</dimIncrement>
          <name>SECCHANNEL[%s]</name>
          <description>no description available</description>
          <addressOffset>0</addressOffset>
          <register>
            <name>PCFG1</name>
            <description>Configuration register 1 for channel pair</description>
            <addressOffset>0xC20</addressOffset>
            <size>32</size>
            <access>read-write</access>
            <resetValue>0</resetValue>
            <resetMask>0x401</resetMask>
            <fields>
              <field>
                <name>PAIRENABLE</name>
                <description>Enable for this channel pair..</description>
                <bitOffset>0</bitOffset>
                <bitWidth>1</bitWidth>
                <access>read-write</access>
              </field>
              <field>
                <name>ONECHANNEL</name>
                <description>Single channel mode.</description>
                <bitOffset>10</bitOffset>
                <bitWidth>1</bitWidth>
                <access>read-write</access>
              </field>
            </fields>
          </register>
          <register>
            <name>PCFG2</name>
            <description>Configuration register 2 for channel pair</description>
            <addressOffset>0xC24</addressOffset>
            <size>32</size>
            <access>read-write</access>
            <resetValue>0</resetValue>
            <resetMask>0x1FF0000</resetMask>
            <fields>
              <field>
                <name>POSITION</name>
                <description>Data Position.</description>
                <bitOffset>16</bitOffset>
                <bitWidth>9</bitWidth>
                <access>read-write</access>
              </field>
            </fields>
          </register>
          <register>
            <name>PSTAT</name>
            <description>Status register for channel pair</description>
            <addressOffset>0xC28</addressOffset>
            <size>32</size>
            <access>read-write</access>
            <resetValue>0</resetValue>
            <resetMask>0xF</resetMask>
            <fields>
              <field>
                <name>BUSY</name>
                <description>Busy status for this channel pair.</description>
                <bitOffset>0</bitOffset>
                <bitWidth>1</bitWidth>
                <access>read-write</access>
              </field>
              <field>
                <name>SLVFRMERR</name>
                <description>Save Frame Error flag.</description>
                <bitOffset>1</bitOffset>
                <bitWidth>1</bitWidth>
                <access>read-write</access>
              </field>
              <field>
                <name>LR</name>
                <description>Left/Right indication.</description>
                <bitOffset>2</bitOffset>
                <bitWidth>1</bitWidth>
                <access>read-write</access>
              </field>
              <field>
                <name>DATAPAUSED</name>
                <description>Data Paused status flag.</description>
                <bitOffset>3</bitOffset>
                <bitWidth>1</bitWidth>
                <access>read-only</access>
              </field>
            </fields>
          </register>
        </cluster>
```

What might be causing the error?

felipebalbi, 2023-08-01:
Here is the full trace output:

```
[DEBUG svd2rust] Current svd2rust config: Config {
        target: CortexM,
        atomics: false,
        atomics_feature: None,
        generic_mod: false,
        make_mod: false,
        const_generic: false,
        ignore_groups: false,
        keep_list: false,
        strict: true,
        pascal_enum_values: false,
        derive_more: false,
        feature_group: false,
        feature_peripheral: false,
        max_cluster_size: false,
        impl_debug: false,
        impl_debug_feature: None,
        output_dir: ".",
        input: Some(
            "svd/MIMXRT633S.svd",
        ),
        source_type: Xml,
        log_level: Some(
            "trace",
        ),
        interrupt_link_section: None,
    }
[INFO  svd2rust] Parsing device from SVD file
[INFO  svd2rust] Rendering device
[DEBUG svd2rust::generate::device] Rendering interrupts
[DEBUG svd2rust::generate::device] Rendering peripheral RSTCTL0
[DEBUG svd2rust::generate::peripheral] Checking derivation information
[DEBUG svd2rust::generate::peripheral] Pushing cluster & register information into output
[TRACE svd2rust::generate::peripheral] Register: SYSRSTSTAT, DeriveInfo: Root
[TRACE svd2rust::generate::peripheral] Register: PRSTCTL0, DeriveInfo: Root
[TRACE svd2rust::generate::peripheral] Register: PRSTCTL1, DeriveInfo: Root
[TRACE svd2rust::generate::peripheral] Register: PRSTCTL2, DeriveInfo: Root
[TRACE svd2rust::generate::peripheral] Register: PRSTCTL0_SET, DeriveInfo: Root
[TRACE svd2rust::generate::peripheral] Register: PRSTCTL1_SET, DeriveInfo: Root
[TRACE svd2rust::generate::peripheral] Register: PRSTCTL2_SET, DeriveInfo: Root
[TRACE svd2rust::generate::peripheral] Register: PRSTCTL0_CLR, DeriveInfo: Root
[TRACE svd2rust::generate::peripheral] Register: PRSTCTL1_CLR, DeriveInfo: Root
[TRACE svd2rust::generate::peripheral] Register: PRSTCTL2_CLR, DeriveInfo: Root
[DEBUG svd2rust::generate::peripheral] Pushing 10 register or cluster blocks into output
[DEBUG svd2rust::generate::peripheral] Expanding registers or clusters into Register Block Fields
[TRACE svd2rust::generate::peripheral] Register: SYSRSTSTAT
[TRACE svd2rust::generate::peripheral] Register: PRSTCTL0
[TRACE svd2rust::generate::peripheral] Register: PRSTCTL1
[TRACE svd2rust::generate::peripheral] Register: PRSTCTL2
[TRACE svd2rust::generate::peripheral] Register: PRSTCTL0_SET
[TRACE svd2rust::generate::peripheral] Register: PRSTCTL1_SET
[TRACE svd2rust::generate::peripheral] Register: PRSTCTL2_SET
[TRACE svd2rust::generate::peripheral] Register: PRSTCTL0_CLR
[TRACE svd2rust::generate::peripheral] Register: PRSTCTL1_CLR
[TRACE svd2rust::generate::peripheral] Register: PRSTCTL2_CLR
[DEBUG svd2rust::generate::device] Rendering peripheral CLKCTL0
[DEBUG svd2rust::generate::peripheral] Checking derivation information
[DEBUG svd2rust::generate::peripheral] Pushing cluster & register information into output
[TRACE svd2rust::generate::peripheral] Register: PSCCTL0, DeriveInfo: Root
[TRACE svd2rust::generate::peripheral] Register: PSCCTL1, DeriveInfo: Root
[TRACE svd2rust::generate::peripheral] Register: PSCCTL2, DeriveInfo: Root
[TRACE svd2rust::generate::peripheral] Register: PSCCTL0_SET, DeriveInfo: Root
[TRACE svd2rust::generate::peripheral] Register: PSCCTL1_SET, DeriveInfo: Root
[TRACE svd2rust::generate::peripheral] Register: PSCCTL2_SET, DeriveInfo: Root
[TRACE svd2rust::generate::peripheral] Register: PSCCTL0_CLR, DeriveInfo: Root
[TRACE svd2rust::generate::peripheral] Register: PSCCTL1_CLR, DeriveInfo: Root
[TRACE svd2rust::generate::peripheral] Register: PSCCTL2_CLR, DeriveInfo: Root
[... 1442 lines cut ...]
[TRACE svd2rust::generate::peripheral] Register: PCFG1
[TRACE svd2rust::generate::peripheral] Register: PCFG2
[TRACE svd2rust::generate::peripheral] Register: PSTAT
[TRACE svd2rust::generate::peripheral] Register: CFG1, DeriveInfo: Root
[TRACE svd2rust::generate::peripheral] Register: CFG2, DeriveInfo: Root
[TRACE svd2rust::generate::peripheral] Register: STAT, DeriveInfo: Root
[TRACE svd2rust::generate::peripheral] Register: DIV, DeriveInfo: Root
[TRACE svd2rust::generate::peripheral] Register: FIFOCFG, DeriveInfo: Root
[TRACE svd2rust::generate::peripheral] Register: FIFOSTAT, DeriveInfo: Root
[TRACE svd2rust::generate::peripheral] Register: FIFOTRIG, DeriveInfo: Root
[TRACE svd2rust::generate::peripheral] Register: FIFOINTENSET, DeriveInfo: Root
[TRACE svd2rust::generate::peripheral] Register: FIFOINTENCLR, DeriveInfo: Root
[TRACE svd2rust::generate::peripheral] Register: FIFOINTSTAT, DeriveInfo: Root
[TRACE svd2rust::generate::peripheral] Register: FIFOWR, DeriveInfo: Root
[TRACE svd2rust::generate::peripheral] Register: FIFOWR48H, DeriveInfo: Root
[TRACE svd2rust::generate::peripheral] Register: FIFORD, DeriveInfo: Root
[TRACE svd2rust::generate::peripheral] Register: FIFORD48H, DeriveInfo: Root
[TRACE svd2rust::generate::peripheral] Register: FIFORDNOPOP, DeriveInfo: Root
[TRACE svd2rust::generate::peripheral] Register: FIFORD48HNOPOP, DeriveInfo: Root
[TRACE svd2rust::generate::peripheral] Register: FIFOSIZE, DeriveInfo: Root
[TRACE svd2rust::generate::peripheral] Register: ID, DeriveInfo: Root
[DEBUG svd2rust::generate::peripheral] Pushing 19 register or cluster blocks into output
[DEBUG svd2rust::generate::peripheral] Expanding registers or clusters into Register Block Fields
[ERROR svd2rust] Error rendering device
```

burrbull, 2023-08-02:
```xml
            <addressOffset>0xC24</addressOffset>
```
When register is in cluster `addressOffset` of register means offset from start of cluster, not start from peripheral.

```xml
<addressOffset>0</addressOffset>
```
`addressOffset` of cluster if offset from start of peripheral.

PS. Wrong repository for issue.

burrbull, 2023-08-22:
@felipebalbi  Could we close the issue?

felipebalbi, 2023-08-22:
Yes, sorry. I should've done it already.

closed 2023-08-22 by felipebalbi