rust-embedded/svdtools #151: Remove empty tag

kossnikita, 2023-08-28:
How to remove empty `fields` tag?

```xml
<peripheral>
      <name>FLASH</name>
      <description>Flash memory controler</description>
      <groupName>FLASH</groupName>
      <registers>
        <register>
          <name>PSR</name>
          <displayName>PSR</displayName>
          <description>Performance selection register</description>
          <addressOffset>0x0</addressOffset>
          <size>0x20</size>
          <resetValue>0x00000030</resetValue>
          <fields>
          </fields>
         </register>
       </registers>
</peripheral>
```

burrbull, 2023-08-28:
I hope you are using rust version of `svdtools`?

kossnikita, 2023-08-28:
> I hope you are using rust version of `svdtools`?

Yes. But I just realized that svdtools does this by default and I don't need to give any instructions.

closed 2023-08-28 by kossnikita