rust-embedded/svdtools #152: Unknown access variant 'read-write ' found

kossnikita, 2023-08-28:
Unnecessary space in access variant `read-write ` results to error

```sh
[2023-08-28T07:35:10Z ERROR svdtools::cli] by svdtools (0.3.0)

    Caused by:
        0: In device `AT32F415xx_v2`
        1: In peripheral `ERTC`
        2: In register `BPR20DT`
        3: Parsing unknown access at 7318:11
        4: unknown access variant 'read-write ' found
```

Also does it make sense to handle such issue for all tags by default? I have to use patches like this:

```
SPI1:
    CTRL2:
      _modify:
        "MDIV3EN ":
          name: MDIV3EN
```

burrbull, 2023-08-28:
I see such issue for first time. Who has generated such awful SVD?

Technically it is possible to trim whitespaces but you need to modify `svd-parser`.
https://github.com/rust-embedded/svd is better place for this issue.

burrbull, 2023-08-28:
@adamgreig I can't transfer the issue. Yet one reason to move repository under @rust-embedded/tools

burrbull, 2023-08-28:
As fast solution I'd recommend to preprocess SVD xml with `regex` to remove unnesessary whitespaces in tags.