rust-embedded/svdtools #156: modifiedWriteValues ignored in patches

kossnikita, 2023-09-21:
#106  added support for modifiedWriteValues but this ignored in patches

```yaml
  SCR:
    _modify:
      IOSB0:
        access: write-only
        modifiedWriteValues: oneToClear
```

```xml
            <field>
              <name>IOSB0</name>
              <description>Set bit 0</description>
              <bitOffset>0</bitOffset>
              <bitWidth>1</bitWidth>
              <access>write-only</access> << it works
            </field>
```

burrbull, 2023-09-21:
https://github.com/stm32-rs/stm32-rs/blob/da9085da570e8934b205115f128500c27ca721fb/peripherals/flash/flash_v1.yaml#L8-L10

kossnikita, 2023-09-21:
Yes, I do it like in this example. Why doesn't this patch work? Doesn't this provide an advantage when used with svd2rust?

burrbull, 2023-09-21:
How looks your patch? Have you added `_W1C`?

kossnikita, 2023-09-21:
Oh. Thank you. This is not at all the behavior I expected.

```yaml
    RLDF:
      _read:
        NotReached: [0, "Downcounter doesn't reached 0x40"]
        Reached: [1, "Downcounter reached 0x40"]
      _W1C:
        Clear: [1, "Clear flag"]
```

```xml
<modifiedWriteValues>zeroToClear</modifiedWriteValues>
              <enumeratedValues>
                <name>RLDFR</name>
                <usage>read</usage>
                <enumeratedValue>
                  <name>NotReached</name>
                  <description>Downcounter doesn't reached 0x40</description>
                  <value>0</value>
                </enumeratedValue>
                <enumeratedValue>
                  <name>Reached</name>
                  <description>Downcounter reached 0x40</description>
                  <value>1</value>
                </enumeratedValue>
              </enumeratedValues>
              <enumeratedValues>
                <name>RLDFW</name>
                <usage>write</usage>
                <enumeratedValue>
                  <name>Clear</name>
                  <description>Clear flag</description>
                  <value>0</value>
                </enumeratedValue>
              </enumeratedValues>
```

burrbull, 2023-09-21:
`oneToClear` is `_W1C`. Not `_W0C`

kossnikita, 2023-09-21:
I get it. In this case, I need W0C.

burrbull, 2023-12-18:
Can we close this, or there still an issue?

kossnikita, 2023-12-18:
I'm not sure. On the one hand, there is a functionality implements this. By the way, it's undocumented #5 . On the other hand, this behavior seems unintuitive.
I would like to add the possibility of such a patch as in my original example.

burrbull, 2023-12-18:
> I would like to add the possibility of such a patch as in my original example.

Reasonable. Could you make PR?

closed 2023-12-18 by burrbull