rust-embedded/svdtools #158: Derive registers regex

kossnikita, 2023-09-26:
Regex in derive patch doesnt work
Derive from registers is not documented by the way.

Works correctly:

```yaml
FLASH:
  _derive:
    UNLOCK2: UNLOCK
    UNLOCK3: UNLOCK
```

Error:

```yaml
FLASH:
  _derive:
    UNLOCK?: UNLOCK
```

```
    Caused by:
        0: Processing device `AT32A403Axx_v2`
        1: According to `FLASH`
        2: Processing peripheral `FLASH`
        3: Deriving register `UNLOCK?` from `String("UNLOCK")`
        4: `Build error: `address_offset` must be initialized
        5: `address_offset` must be initialized
```

Ignored:

```yaml
FLASH:
  UNLOCK?: 
    _derivedFrom: UNLOCK
```

burrbull, 2023-09-26:
> Regex in derive patch doesnt work

It should not.
If target register exists it removes its contents (fields, etc.) and replaces with "derivedFrom" tag.
If target register does not exist it creates new one.

kossnikita, 2023-09-26:
Why can't it be a regex match else create a new register?
Even if this shouldn't, is it possible to implement it like the last example?

burrbull, 2023-09-26:
try:
```yaml
FLASH:
  _modify:
    UNLOCK?: 
      derivedFrom: UNLOCK
```

kossnikita, 2023-09-26:
`try`
Ignored

kossnikita, 2023-09-26:
Work correctly #159

closed 2023-12-18 by burrbull