rust-embedded/svdtools #161: Panic in field array dim calculation

kossnikita, 2023-10-02:
```yaml
EXINT:
  INTEN:
    _array:
      INTEN?,INTEN10:
        description: "Interrupt enable or disable on line %s"
```
Token boundaries are calculated without comma split.

https://github.com/rust-embedded/svdtools/blob/27cf2b136627e7fad154b66d4295b10e3fa02682/src/patch/register.rs#L361

```
li = 5, ri = 8
thread 'main' panicked at 'byte index 18446744073709551614 is out of bounds of `INTEN0`', src\patch\register.rs:383:30
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

https://github.com/rust-embedded/svdtools/blob/27cf2b136627e7fad154b66d4295b10e3fa02682/src/patch/register.rs#L383

burrbull, 2023-10-02:
I know about this issue, but I don't know good way to fix it.

burrbull, 2023-10-02:
#161 should help for your specific situation, but this is just a patch.

burrbull, 2023-10-02:
Although this can break other use cases.

kossnikita, 2023-10-02:
Yes, works in this case

```yaml
EXINT:
  INTEN:
    _array:
      INTEN?,INTEN1[0-7]:
        description: "Interrupt enable or disable on line %s"
```

```xml
<fields>
  <field>
    <dim>18</dim>
    <dimIncrement>0x1</dimIncrement>
    <dimIndex>0-17</dimIndex>
    <name>INTEN%s</name>
    <description>Interrupt enable or disable on line %s</description>
```

But don't work there (I don't know why someone would write such a weird patch)

```yaml
EXINT:
  INTEN:
    _array:
      INTEN1[0-7],INTEN?:
        description: "Interrupt enable or disable on line %s"
```

```xml
<fields>
  <field>
    <dim>18</dim>
    <dimIncrement>0x1</dimIncrement>
    <dimIndex>,,,,,,,,,,0,1,2,3,4,5,6,7</dimIndex>
    <name>INTEN1%s</name>
    <description>Interrupt enable or disable on line %s</description>
```

burrbull, 2023-10-02:
I don't have better ideas.

P.S. We need someone to improve [YAML Format](https://github.com/rust-embedded/svdtools#device-and-peripheral-yaml-format) as patch format was much extended since last docs edition. Do you have a wish to help? I'm bad in doc writing.

kossnikita, 2023-10-02:
As far as I know, the functionality of rust and python is now slightly different. Should this be reflected in the documentation or should the new syntax be introduced into Python first?

burrbull, 2023-10-02:
Python version is deprecated, but we try to support it if it is not hard. For example #162 is easy, but I can't say the same for #159. For now differences are not big.

PR #162 (2023-10-02): allow , in spec_ind
Fix #161
closed 2023-10-02 by burrbull