rust-embedded/svdtools #174: M302: Size of Register must be 8, 16 or 32 Bits

kossnikita, 2023-10-25:
From [CMSIS SVDConv docs](https://arm-software.github.io/CMSIS_5/SVD/html/svd_SVDConv_pg.html#:~:text=Size%20of%20Register%20%27NAME%27%3A%27NUM%27%20must%20be%208%2C%2016%20or%2032%20Bits)

<html>
<body>

Message Number | Type | Message Text 
-- | -- | --
M302 | ERROR | Size of Register 'NAME':'NUM' must be 8, 16 or 32 Bits

</body>
</html>

This limitation is not obvious to me. For example, it is easier to create a 12-bit register for a 12-bit ADC than a 32-bit register with only one field.
svd2rust handles this by creating an unsafe method for working with such registers. But CMSIS prohibits such svd files. Should utilities handle this error?

burrbull, 2023-10-25:
> This limitation is not obvious to me. For example, it is easier to create a 12-bit register for a 12-bit ADC than a 32-bit register with only one field.

Register is how CPU reads/writes memory. It does it by u8,u16,u32, u64 depending on architecture. Even if it physically only 12 lines are connected to register mapped on memory.
Field is abstraction which you can push/pop from register.

kossnikita, 2023-10-25:
Do you mean that there are modes in which the processor can read or write registers differently? I thought that the data bus always has a fixed width. For example, a 32bit processor will always read/write all 32bits.

burrbull, 2023-10-25:
> Do you mean that there are modes in which the processor can read or write registers differently?

Yes, depending on asm command you are using to read/write register microprocessor can react differently.

Usually this is allowed for data registers, for example SPI DR, TIM ARR, etc. This is what `<size>` tag about.

[image]

kossnikita, 2023-10-25:
Thank you, it's clear now. So should introduce this limitation into the svdtool or into the svd?