rust-embedded/svdtools #189: 0.3.6 doesn't patch a clustered register as 0.2.8 did

SpinFast, 2023-12-17:
My svd has a peripheral PWM with a cluster SM and register SMOCTRL

In 0.2.8 I could patch fields in SMOCTRL with...

PWM?:
  SMOCTRL:
    _clear:
      - PWMXFS
      - PWMBFS
      - PWMAFS

With 0.3.6 it can't find the SWOCTRL register and its unclear to me how to select it (in the cluster SM)

Actual yaml patch
https://github.com/imxrt-rs/imxrt-ral/blob/master/devices/imxrt1176_cm7.yaml#L105

Actual SVD
https://github.com/imxrt-rs/imxrt-ral/blob/master/svd/imxrt1176_cm7.svd

burrbull, 2023-12-18:
Could you try to find first breaking release/PR?

What exactly is the error?
Also with last release you could run
```
svdtools patch --show-patch-on-error your_patch.yaml
```
to see full patch with all `_include`s merged.

burrbull, 2023-12-18:
Oh. I see. It is https://github.com/rust-embedded/svdtools/issues/144 related.
Forgot about it.
Should be easy to fix.

burrbull, 2023-12-18:
> With 0.3.6 it can't find the SWOCTRL register and its unclear to me how to select it (in the cluster SM)

Should be something like this. Although I've never used it.
```yaml
PWM?:
  _clusters:
    "SM[%s]":
      SMOCTRL:
        _clear:
          - PWMXFS
          - PWMBFS
          - PWMAFS
```

SpinFast, 2023-12-19:
When I try this with 0.3.6

```yaml
PWM?:
  _clusters:
    "SM[%s]":
      SMOCTRL:
        _clear:
          - PWMXFS
          - PWMBFS
          - PWMAFS
```

I get

```shsvdtools patch devices/imxrt1176_cm7.yaml
[2023-12-19T04:23:51Z ERROR svdtools::cli] by svdtools (0.3.6)
    
    Caused by:
        0: Processing device `MIMXRT1176_cm7`
        1: According to `PWM?`
        2: Processing peripheral `PWM1`
        3: According to `SM[%s]`
        4: Could not find `PWM1:SM[%s]. Present clusters: SM[%s].`
```

That's pretty confusing to me I have to say!

burrbull, 2023-12-19:
Try to escape brackets:
`SM\[%s\]`

Logiase, 2024-01-12:
Hi @burrbull @SpinFast 

`SM\[%s\]` only works on Linux, but failed on Windows with same version 0.3.8

burrbull, 2024-01-12:
Looks like we need to enable `backslash_escape` feature. See https://docs.rs/globset/latest/globset/#syntax.

Make PR, please, if you can.

closed 2024-01-13 by Emilgardis

cosmikwolf, 2024-09-24:
Just wanted to pop in here to say to get this to work on macos, i had to double escape the brackets:

```
I2C0:
  _clusters:
    "I2C0_GPRCM\\[%s\\]":
      _strip:
        - "I2C0*"
```

PR #197 (2024-01-13): escape on match
cc @Logiase @Emilgardis 

closes #189