rust-embedded/svdtools #202: Making a cluster array of register arrays

ejpcmac, 2024-02-07:
## Context

Let’s say we have some peripheral that exposes an interface to feed it with some 128-bit inputs (let’s say 3 for the example), and we have a 32-bit bus. It then provides registers like: `INPUT0A`, `INPUT0B`, `INPUT0C`, `INPUT0D`, `INPUT1A`, ..., `INPUT3D`. `INPUT0A` is the first 32-bit word in input 0, `INPUT3B` the second 32-bit word in input 3, and so on.

As you can imagine, working with such registers in a generic way in the code is not very handy, so I’d like to group them in some kind of two-dimentional array, like `INPUT[n][w]`, where `n` is the input number and `w` the 32-bit word index inside this 128-bit input.

## Trials with `svdtools`

The first step I’ve done is to create arrays for each input, like:

```yml
PERIPHERAL:
    _array:
        INPUT0?: {}
        INPUT1?: {}
        INPUT2?: {}
```

This is better since I can now do in my Rust code:

```rust
peripheral.input0[w].write(|w| ...);
// ...
peripheral.input1[w].write(|w| ...);
```

There is still however a hard-coded index in the names, so I need to use macros to get generic code, which is not optimal.

## Trials with the SVD file

While looking to the SVD documentation, I came to see that we can put registers in clusters, and we can do arrays of clusters. I can then write something like:

```xml
<cluster>
  <dim>3</dim>
  <dimIncrement>0x10</dimIncrement>
  <dimIndex>0-2</dimIndex>

  <name>INPUT%s</name>
  <description>Input registers</description>
  <addressOffset>0x100</addressOffset>

  <register>
    <dim>4</dim>
    <dimIncrement>0x04</dimIncrement>
    <dimIndex>0-3</dimIndex>

    <name>WORD%s</name>
    <description>Word</description>
    <addressOffset>0x0</addressOffset>
    <size>32</size>
    <access>read-write</access>
    <resetValue>0x0</resetValue>
    <resetMask>0xFFFFFFFF</resetMask>

    <fields>
      <field>
        <name>WORD</name>
        <description>Word</description>
        <bitOffset>0</bitOffset>
        <bitWidth>32</bitWidth>
        <access>read-write</access>
      </field>
    </fields>
  </register>
</cluster>
```

This allows me to write Rust code like:

```rust
peripheral.input[n].word[w].write(|w| ...);
```

which is exactly what I want.

## How to do this with `svdtools`?

Is it possible to achieve this with `svdtools`? I see two possible paths:

1. Is there a way to create clusters from register arrays, and then an array from clusters? I’ve tried to do so but I’m getting some errors.
2. Is there a way to `_add` clusters and arrays? (Actually the original SVD I’m using has the registers for only 1 input, so I’d be fine to just define them from scratch in a `_add` block)

burrbull, 2024-02-07:
2nd should be possible with:
```yaml
_add:
  _clusters:
    DATA
```

ejpcmac, 2024-02-07:
I am indeed able to write:

```yml
_add:
    _clusters:
        INPUT:
            description: "Input registers"
            addressOffset: 0x100
            registers:
                WORD:
                    description: Word
                    addressOffset: 0x0
                    access: read-write
                    resetValue: 0
                    fields:
                        WORD:
                            description: Word
                            bitOffset: 0
                            bitWidth: 32
```

However, this gives me a cluster containing a register, not an array of clusters containing an array of registers. I can’t figure how to create an array from scratch.

I tried as well to specify the `dim` keys inline:

```yml
_clusters:
    INPUT%s:
        dim: 6
        dimIncrement: 0x10
        dimIndex: 0-5
        description: "Input registers"
        addressOffset: 0x100
        registers:
            WORD:
                dim: 4
                dimIncrement: 0x04
                dimIndex: 0-3
                description: Word
                addressOffset: 0x0
                size: 32
                access: read-write
                resetValue: 0
                fields:
                    WORD:
                        description: Word
                        bitOffset: 0
                        bitWidth: 32
```

While they are preserved in the cluster, they are removed from the register.

burrbull, 2024-02-07:
> While they are preserved in the cluster, they are removed from the register.

This is strange.

burrbull, 2024-02-07:
I've found. The issue is here:
https://github.com/rust-embedded/svdtools/blob/cc456bc2702fb1d07c4a9fd4e5d54dd83f868ee8/src/patch/mod.rs#L531-L535

There should be something like there:
https://github.com/rust-embedded/svdtools/blob/cc456bc2702fb1d07c4a9fd4e5d54dd83f868ee8/src/patch/peripheral.rs#L456-L461

PR #212 (2024-03-06): cluster of arrays
Closes #202 

cc @ejpcmac
closed 2024-03-06 by burrbull