rust-embedded/svdtools #207: How to delete the `dim` register?

wsndshx, 2024-02-18:
I have registers that need to be deleted as follows:

```xml
        <register>
          <dim>2</dim>
          <dimIncrement>0x4</dimIncrement>
          <name>gpio_set[%s]</name>
          <description>Set pin output value to high</description>
          <addressOffset>0xAEC</addressOffset>
        </register>
```

However, I attempted to use `_delete` without success:

```yaml
  _delete:
    gpio_set[%s]:
      dim: 35
```

or

```yaml
  _delete:
    gpio_set:
      dim: 35
```

At the same time, I also tried using `modify` to make changes, but it didn't work either:

```yaml
  _modify:
    gpio_set[%s]:
      dim: 35
```

or

```yaml
  _modify:
    gpio_set:
      dim: 35
```

It seems like I cannot select this register. How can I do it?

burrbull, 2024-02-18:
> It seems like I cannot select this register. How can I do it?

 Brackets are specsymbols. Try:

```yaml
  _modify:
    gpio_set\[%s\]:
      dim: 35
```

wsndshx, 2024-02-18:
> > It seems like I cannot select this register. How can I do it?
> 
> Brackets are specsymbols. Try:
> 
> ```yaml
>   _modify:
>     gpio_set\[%s\]:
>       dim: 35
> ```

Great, it worked. Thank you.

closed 2024-02-18 by wsndshx