rust-embedded/svdtools #230: Request to update lxml version dependency information

taotieren, 2024-05-29:
I'm having problems with the lxml version dependency in the [python-svdtools](https://aur.archlinux.org/packages/python-svdtools) package that maintains the AUR.

```bash
...
==> WARNING: Skipping all source file integrity checks.
==> Extracting sources...
-> Extracting svdtools-0.3.9.tar.gz with bsdtar
==> Entering fakeroot environment...
==> Starting package_python-svdtools()...
* Getting build dependencies for wheel...

ERROR Missing dependencies:
lxml~=4.6
==> ERROR: A failure occurred in package_python-svdtools().
Aborting...
...
```

`lxml` relies on the **4.6** version of the package, which is `python2-lxml` by checking arch's packages. now arch only supports python3 which requires the `python-lxml` package, but the `python-lxml` version is **5.2.2**. But the `python-lxml` version is **5.2.2**. It does not meet the dependency requirements of svdtools, please upgrade the lxml version dependency information of svdtools.

```bash
yay -Ss python-lxml
...
aur/python2-lxml 4.9.4-2 (+10 0.60)
Python2 binding for the libxml2 and libxslt libraries
blackarch/python2-lxml 4.6.2-2 (1.2 MiB 4.4 MiB)
Powerful and Pythonic XML processing library combining libxml2/libxslt with the ElementTree API.
...
extra/python-lxml 5.2.2-1 (1.3 MiB 5.0 MiB) (已安装: 5.2.2-1.1)
Python3 binding for the libxml2 and libxslt libraries
extra/python-lxml-docs 5.2.2-1 (3.0 MiB 10.9 MiB)
Python binding for the libxml2 and libxslt libraries (docs)
...
```

burrbull, 2024-05-29:
Open PR, please.
And note that python `svdtools` is deprecated and can be removed soon. The main line is rust based version of `svdtools`.
See https://crates.io/crates/svdtools or https://github.com/rust-embedded/svdtools/releases

taotieren, 2024-05-30:
Thanks, I'll synchronize the package change processing on my end.