rust-embedded/svdtools #239: v0.3.17 fails to build on 1.80 with `--locked`

newAM, 2024-08-10:
`cargo install svdtools --locked` with rustc v1.80

Results in:

```
error[E0282]: type annotations needed for `Box<_>`
  --> /home/alex/.cargo/registry/src/index.crates.io-6f17d22bba15001f/time-0.3.31/src/format_description/parse/mod.rs:83:9
   |
83 |     let items = format_items
   |         ^^^^^
...
86 |     Ok(items.into())
   |              ---- type must be known at this point
   |
help: consider giving `items` an explicit type, where the placeholders `_` are specified
   |
83 |     let items: Box<_> = format_items
   |              ++++++++
```

Would it be possible to get a crates.io release to update the lockfile?  I maintain the NixOS package for svdtools, and NixOS builds with the release lockfile.

burrbull, 2024-08-10:
I do not like the idea to add `Cargo.lock` to master branch.

But it makes sense to add it in releases (on `crates.io` and Github Releases). Could it be done by some `deploy CI`?
cc @Emilgardis 
Related RP: #204 

@newAM I want merge #238 before release. But it is blocked by https://github.com/stm32-rs/stm32-rs/pull/1002

Emilgardis, 2024-08-10:
A lockfile will always be provided when `cargo publish`ing, so nothing needs to be changed really

burrbull, 2024-08-10:
@newAM released.

closed 2024-08-10 by newAM