rust-embedded/svdtools #256: There is no way to define enumerated values

bugadani, 2024-11-20:
The register -> field -> enumeratedValues key is ignored.

burrbull, 2024-11-20:
`svdtools` **was created** to add `enumeratedValues`. See [README](https://github.com/rust-embedded/svdtools?tab=readme-ov-file#device-and-peripheral-yaml-format)

bugadani, 2024-11-20:
I see, so the expected workflow when defining new peripherals is, to first add the peripheral without enumerated values, then modify it immediately? I was hoping to do something like the following, since most of the other properties work like this:

```yaml
  field:
    description: "foo"
    ...
    enumeratedValues:
      ...
```

closed 2024-11-20 by bugadani

burrbull, 2024-11-20:
> I was hoping to do something like the following, since most of the other properties work like this:

You could add this possibility here: https://github.com/rust-embedded/svdtools/blob/a24feb52b961bc9903e8114e18787932b23b6585/src/patch/mod.rs#L477