rust-embedded/svdtools #273: Share identical field definitions between separate clusters

astapleton, 2025-01-30:
If a a peripheral is partitioned into multiple clusters because registers differ on a subset of the clustered registers between partitions, it would be nice to be able to share the enum definitions from registers that are identical. Even better would be to also share the enum definitions in identical fields in registers that have extra fields.

As an example, the STM32H5xx series GPDMA peripheral has a bank of 8 channels. [Channels 0-5](https://docs.rs/stm32h5-staging/0.18.1/stm32h5_staging/stm32h503/gpdma1/ch/index.html) are identical, but [channels 6 & 7](https://docs.rs/stm32h5-staging/0.18.1/stm32h5_staging/stm32h503/gpdma1/ch2d/index.html) have extra functionality, which manifests in extra fields in a few of those registers. Unfortunately this means that the bank is partitioned into channels 0-5 and channels 6-7 and all the identical field definitions are duplicated across the separate partitions, which also means that a single common enum cannot be used for addressing identical registers across all 8 channels.

burrbull, 2025-01-31:
It is already possible with fully-qualified paths but should be easier with https://github.com/rust-embedded/svdtools/pull/274

The more valuable issue is `svd2rust` https://github.com/rust-embedded/svd2rust/issues/903

astapleton, 2025-01-31:
Ah, great. I was trying something with register patches before the cluster step. I didn't know you could put `_derives` inside the _cluster definitions. I'll try this when [#904](https://github.com/rust-embedded/svd2rust/pull/904) lands.

closed 2025-01-31 by astapleton