rust-embedded/svdtools #279: support for <enumeratedValues> from CMSIS-SVD specification

vinchatl, 2025-02-09:
Even though this features is not always populated by devices manufacturers, enumerated values are a [complete part of the CMSIS-SVD specification](https://arm-software.github.io/CMSIS_5/SVD/html/svd_Format_pg.html).

Would it be possible to support them to create crates ?

For example, here's an extract of the STM32U535 SVD (v1.0) for FLASH_OPTR register:

```xml

            <field>
              <name>RDP</name>
              <description>Readout protection level
Others: Level 1 (memories readout protection active)
Note: Refer to  for more details.</description>
              <bitOffset>0</bitOffset>
              <bitWidth>8</bitWidth>
              <access>read-write</access>
              <enumeratedValues>
                <enumeratedValue>
                  <name>B_0xAA</name>
                  <description>Level 0 (readout protection not active)</description>
                  <value>0xAA</value>
                </enumeratedValue>
                <enumeratedValue>
                  <name>B_0x55</name>
                  <description>Level 0.5 (readout protection not active, only non-secure debug access is possible). Only available when TrustZone is active (TZEN=1)</description>
                  <value>0x55</value>
                </enumeratedValue>
                <enumeratedValue>
                  <name>B_0xCC</name>
                  <description>Level 2 (chip readout protection active)</description>
                  <value>0xCC</value>
                </enumeratedValue>
              </enumeratedValues>
            </field>
```

and the associated documentation from the RM:

[image]

The `<name>` is a bit weird (one would imagine `"Level2"`, `"Level1"`, `"Level0.5"` and `"Level0"`), but the `<value>` and `<description>` are basically what's expected to create a well documented crate.

burrbull, 2025-02-09:
> Would it be possible to support them to create crates ?

It was possible from first day.

`B_0xCC` is definitely weird. That is why we remove them all (if already present) and then add with fine names.
See  https://github.com/stm32-rs/stm32-rs/tree/master/devices/fields for examples.

Removed enums I've added in this branch: https://github.com/stm32-rs/stm32-rs/tree/enums/enums
So you can copy parts you need from there.

closed 2025-02-09 by burrbull

vinchatl, 2025-02-09:
That makes sense, but it's a bit sad that the enum values are badly named by manufacturer (ST in our case), it could have avoid a lot of work.

Thanks for the links.